NordicSemiconductor/pc-ble-driver-py #3: Request for more advanced examples

ksachdeva, 2016-06-29:
Hi,

Tried the examples and they do work as you mentioned in the post however these are very basic examples. Since there is no documentation / guide on how to use python library would appreciate if you could provide more examples. For e.g. I am looking for the ability to

* issue scan of devices (your heart rate collector does this)
* issue discovery of services (your heart rate kind of does this)
* I want my app to be notified when the service discovery is done
* Once discovery is done I want to have the handles of the characteristics of various services. is there an event/callback that is fired when discovery process is done !!
* Once I have the handle of the desired characteristics I want to perform read and write operations

Would really appreciate an example like this.

Regards & thanks 
Kapil

carlescufi, 2016-06-29:
This is an initial release with only a subset of all the functionality abstracted in BLEDriver and BLEAdapter, but the whole of the SoftDevice API is available to Python apps in the pc_ble_driver SWIG module. We encourage users to contribute their own application examples and enhancements to the libraries so that they can  build up over time, and Nordic engineers will also make contributions in the future.  For this first release we have focused on the features that are  necessary to perform Device Firmware Update.

bihanssen, 2017-04-12:
Closing the issue due to inactivity.

closed 2017-04-12 by bihanssen