NordicSemiconductor/pc-ble-driver-py #23: Invalid BLEGapRoles value upon connect in heart_rate_collector.py

IvoDoppler, 2017-06-01:
I'm not able to connect to a peripheral using the `heart_rate_collector.py` example.

After enabling logging, I see I'm getting random values upon `self.adapter.connect(peer_addr)`:

```
ERROR:pc_ble_driver_py.ble_driver:Exception: 49 is not a valid BLEGapRoles
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\pc_ble_driver_py\\ble_driver.py', 1326, 'sync_ble_evt_handler', 'role           = BLEGapRoles(connected_evt.role),')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 347, '__call__', 'return cls.__new__(cls, value)')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 662, '__new__', 'raise ValueError("%s is not a valid %s" % (value, cls.__name__))')
ERROR:pc_ble_driver_py.ble_driver:
ERROR:pc_ble_driver_py.ble_driver:Exception: 192 is not a valid BLEHci
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\pc_ble_driver_py\\ble_driver.py', 1335, 'sync_ble_evt_handler', 'reason      = BLEHci(disconnected_evt.reason))')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 347, '__call__', 'return cls.__new__(cls, value)')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 662, '__new__', 'raise ValueError("%s is not a valid %s" % (value, cls.__name__))')
```

(the actual values for BLEGapRoles and BLEHci differ in every run)

bencefr, 2017-06-01:
Could you give us a bit more info on this? I just tried the example and it was working fine.
Did you modify the example code? What are your devices/environment that could be relevant?

IvoDoppler, 2017-06-01:
I only changed `TARGET_DEV_NAME`, and added `logging.basicConfig()` to enable logging.
I running the example on Microsoft Windows 10 Home (version 1703, build 15063.332) 64-bit with a nRF51 USB dongle, and testing against a Tile.

Even without a BLE peripheral, I get a similar error when discovery times out:
```
ERROR:pc_ble_driver_py.ble_driver:Exception: 192 is not a valid BLEGapTimeoutSrc
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\pc_ble_driver_py\\ble_driver.py', 1374, 'sync_ble_evt_handler', 'src          = BLEGapTimeoutSrc(timeout_evt.src))')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 347, '__call__', 'return cls.__new__(cls, value)')
ERROR:pc_ble_driver_py.ble_driver:('C:\\Python27\\lib\\site-packages\\enum\\__init__.py', 662, '__new__', 'raise ValueError("%s is not a valid %s" % (value, cls.__name__))')
```

bencefr, 2017-06-02:
@IvoDoppler , while we try to get to the bottom of this, could you maybe try fix_swig_ble_event_memory_alloc branch if that fixes the issue for you?

IvoDoppler, 2017-06-02:
@bencefr on the fix_swig_ble_event_memory_alloc branch, time out no longer generates an invalid event exception:
```
Traceback (most recent call last):
  File "heart_rate_collector.py", line 188, in <module>
    main(serial_port)
  File "heart_rate_collector.py", line 152, in main
    conn_handle = collector.connect_and_discover()
  File "heart_rate_collector.py", line 88, in connect_and_discover
    new_conn = self.conn_q.get(timeout = 60)
  File "C:\Python27\lib\Queue.py", line 176, in get
    raise Empty
Queue.Empty
```

But when a target device is found, I'm not getting any connection event at all.

bencefr, 2017-06-06:
@IvoDoppler , the python driver is hardcoded to use the connectivity firmware with baud rate 115200. If you have used e.g. nRF Connect desktop application with your dongle, that might have flashed the faster version with baud rate 1000000. As I see the code only checks if the version of the firmware matches, not the speed of it, therefore serial communication will fail.

Until better example or solution is provided, you can either erase the donlge `nrfjprog -e -f NRF51`, and let the driver flash again, or leave the fast firmware on the dongle and change the source in heart_rate_collector.py[146]:
```
   driver    = BLEDriver(serial_port=serial_port, baud_rate=1000000, auto_flash=True)
```
I hope this helps.

IvoDoppler, 2017-06-06:
After specifying baudrate of 1000000, I was not able at all, and got the following error:
```
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13
```

Nowever, `nrfjprog -e -f NRF51` did fix the issue.

closed 2017-06-06 by IvoDoppler