NordicSemiconductor/pc-ble-driver-py #24: Exception: <value> is not a valid BLEGapTimeoutSrc

metc, 2017-06-13:
I am using an nRF51 dongle in order to print all BLE devices advertising around me. I got the following error when the `gap_evt_timeout` is handle by the Python driver:

```
15:48:03: Start scanning...
15:48:03: Received ADV report, address: 0x06829735AABB, device_name: Nordic_1
15:48:08: Received ADV report, address: 0x06829735BBCC, device_name: Nordic_1
15:48:08: Exception: 227 is not a valid BLEGapTimeoutSrc
15:48:08: ('C:\\Python279-32\\lib\\site-packages\\pc_ble_driver_py\\ble_driver.py', 1374, 'sync_ble_evt_handler', 'src          = BLEGapTimeoutSrc(timeout_evt.src))')
15:48:08: ('C:\\Python279-32\\lib\\site-packages\\enum\\__init__.py', 347, '__call__', 'return cls.__new__(cls, value)')
15:48:08: ('C:\\Python279-32\\lib\\site-packages\\enum\\__init__.py', 662, '__new__', 'raise ValueError("%s is not a valid %s" % (value, cls.__name__))')
```

The GAP ADV timeout of 10 seconds is correct. I use the following code on an nRF51 dongle which is flashed automatically using the library (pc_ble_driver_sd_api_v2):

```c
scan_params = BLEGapScanParams(interval_ms=200, window_ms=150, timeout_s=10)
self.adapter.driver.ble_gap_scan_start(scan_params)
```

I am not sure how to debug this. Could you help me? The value of the `BLEGapTimeoutSrc` changes for every run.

bencefr, 2017-06-14:
@metc , I have stripped down the heart_rate_collector example to do only discovery and nothing else, and it is working just fine with adding your scan_params.
Reading the source BLEGapTimeoutSrc is called by an Enum that has range of [0..4] in value defined in pc-ble-driver, instead this you get random value, which makes me suspect that different versions of structs mapped to the BLEEvent is used here. In short, my suggestion - since I don't know more about your code - is:
1. try to erase the device to get it flashed again `nrfjprog -e -f nrf51`
2. reinstall pc-ble-driver-py
3. step 2 followed by step 1

mrodem, 2017-06-14:
I was able to reproduce the issue when running the heart rate collector example with the nRF51 dongle. It was not immediately obvious, because logging was disabled in the example. I added `logging.basicLogging()` to the top of the example, and then the `Exception: <value> is not a valid BLEGapTimeoutSrc` was printed after a few seconds.

The issue seems to be related to d9e06aa4e which was added in version 0.8.1. We will have to look into this and apply a fix. I have tested version 0.8.0, and that seems to be working fine with the nRF51 dongle, so you could perhaps use that version in the meantime.

bencefr, 2017-06-15:
#25 should fix this

closed 2017-06-15 by bencefr