NordicSemiconductor/pc-ble-driver-py #32: Installing on macOS

eblot, 2017-08-17:
The installation instructions recommend to use brew to build (as Unix dependencies such as *cmake* and *swig* are required).

The installation may fail if a more recent python 2.x version has been installed with brew, even if a virtual env is used, because the system mixes libraries from the Apple installation (from `/System/Library/Frameworks/Python.framework/Versions/2.7/Python`) and from the brew installation (from `/usr/local/Cellar/python/2.7.*/Frameworks/Python.framework/Versions/2.7/Python`).

When the simplest test program is started, the Python VM fails with the ciphered error message `Fatal Python error: PyThreadState_Get: no current thread`

You need to (re)build the pc-ble-driver library with the system Python VM.

brew does not install boost libraries with universal builds. If you do not want to mess your system with an old boost library and that you run a 64 bit host - which should be most of the macOS installations nowadays - install boost normally with brew, then edit `apple.cmake` file and remove 32 bit support:

    set(CMAKE_OSX_ARCHITECTURES "x86_64")

It would be nice to see all those tools, instructions, everything updated. 
GCC 4.9 (to build nRF52 SDK) is outdated, 32-bit PC installations are outdated, Python SWIG-based wrappers are outdated, Boost 1.55 is nearly 4 years old, Python 2.x is outdated etc, etc. What a pity nRF52 is a new product, but relies on such outdated tools.

ThomasGerstenberg, 2017-09-21:
I ran into the same issue on macOS with brewed python. Confirmed that rebuilding worked for me, but I had to specify `-DPYTHON_LIBRARY=$BREW_PYTHON_LOC` to cmake.

There's another fix that works which doesn't require rebuilding the libraries, see the devzone answer [here](https://devzone.nordicsemi.com/question/87698/nrfutil-on-osx/?answer=153831#post-id-153831).  It uses `install_name_tool` to change the dependent python library in the `_pc_ble_driver_sd_api_v*.so` to point to the brewed python

bihanssen, 2019-01-29:
Thanks for reporting, comments have been noted. Closing this issue.

closed 2019-01-29 by bihanssen