NordicSemiconductor/pc-ble-driver-py #35: BLEHci.conn_failed_to_be_established before service_discovery

kromancer, 2017-09-06:
Hello guys and keep up the good work!

I want to ask your opinion about  how to handle a disconnection event after a successful connection and before service discovery. I am using the pc-ble-driver-py version 0.9.1 to connect to a peripheral.

To be more specific, let us assume the following procedure:
```python
self.adapter.driver.ble_gap_scan_start()
new_conn = self.conn_q.get(timeout=60)
self.adapter.service_discovery(new_conn)
```

In my setup I sometimes observe the aforementioned phenomenon.
```
...
New connection, conn_handle: 0
Disconnected, conn_handle: 0, reason: BLEHci.conn_failed_to_be_established
Traceback (most recent call last):
...
File "C:\Python27\lib\site-packages\wrapt\wrappers.py", line 562, in __call__
   args, kwargs)
File "C:\Python27\lib\site-packages\pc_ble_driver_py\ble_driver.py", line 124, in wrapper
    err_code = wrapped(*args, **kwargs)
File "C:\Python27\lib\site-packages\pc_ble_driver_py\ble_adapter.py", line 176, in service_discovery
    if response['status'] == BLEGattStatusCode.success:
TypeError: 'NoneType' object has no attribute '__getitem__'
```

Shall I just try to catch the TypeError connection and then retry to connect?

Thank you for your time!

bihanssen, 2017-09-07:
Hi kromancer!
The connect followed by immediate disconnect is unfortunately a scenario that might happen. The code in ble_adapter is not robust against these errors, so your best option is probably to catch the error and try to reconnect.

closed 2017-09-07 by bihanssen

p-sousa, 2022-04-13:
@bihanssen sorry for asking this here in an issue from 2017
Can you please give more details on the reason for these to happen?
I understand the framework is not robust enough to deal with this, but what's the root cause for this? Just a normal scenario of BLE or is it some known issue from pc-ble-driver?

Thank you!

bihanssen, 2022-04-20:
@p-sousa  I can't recall the details, but my understanding is that the origin of the error is from the SoftDevice link layer. Whether ble-driver does anything to provoke the error to happen more often I'm not sure of. Here is a related DevZone article: https://devzone.nordicsemi.com/f/nordic-q-a/44756/how-to-minimize-hci-error-0x3e-and-0x22-connection-failures-of-ble-connections