NordicSemiconductor/pc-ble-driver-py #38: Registering a Custom 128-bit UUID thru BLEAdapter

Darrew, 2017-09-20:
Question> 
Has the adapter a way (method?) to register a custom UUID? In 0.9.1, I do not see any. 

Issue> 
Using driver.ble_vs_uuid_add() seems to work for me (able to discover custom Services and Chars); but after adapter.service_discovery(conn, myuuid) with myuuid an instance of BLEUUID with custom BLEUUIDBase, the object a=adapter.db_conns[conn].services[x].uuid.base returns an instance of BLEUUIDBase but attribute a.base is None. In all, does the driver update a.base at all?

fouge, 2018-10-09:
Same issue here. I'm trying to subscribe to custom chars without success.
Any news regarding that issue?

pambardekar, 2019-04-19:
@Darrew Were you able to discover custom services and characteristics (adapter.service_discovery(conn, None))? I am trying it but I only get default services like battery level.

pambardekar, 2019-04-25:
After adding vendor specific UUID I was able to discover custom 128 bit characteristics.

ilw, 2019-05-02:
I'm unable to interact (get notifications, read characteristics etc) with any VS UUIDs. 

Service discovery seems to find all the handles and attribute types but doesn't get the UUIDs.

pambardekar, 2019-05-02:
After I added a call to ble_vs_uuid_add(), UUIDs started getting discovered correctly for me.

bihanssen, 2020-09-15:
Demonstration of use of 128 bit UUIDs can be found in test_server_client.py, added in PR #172.
Closing this issue. Reopen if needed.

closed 2020-09-15 by bihanssen