NordicSemiconductor/pc-ble-driver-py #42: Reject Connection Parameters Update Request

rpelc, 2017-10-04:
Hello NordicSemiconductor team,

Is there a way you could add option for rejection connection parameters update request from device?
I'm trying to use nrfutil for dfu otau update and it takes forever with connection parameters reqeuested by device.

bihanssen, 2017-10-04:
Hi,
rejecting the connection params update is usually not an option, since the peripheral would usually then terminate the connection. However, it is possible to respond to an update by updating to the lowest connection interval in the requested range, instead of giving the full range and leaving it to the SoftDevice to decide. Are you able to see what the requested range was (min-max conn int), and what the resulting new connection interval was set to?

rpelc, 2017-10-04:
Yes, i can see the minimum value reqested by device, but it is still to high and otau lasts for very long.
I was using nrf connect and rejection of the parameters worked fine - connection was established with lower minimum interval than requested by device. Unfortunately i need to do it from command line and nrf connect doesn't support it.

bihanssen, 2019-01-29:
Closing as won't fix. Please reopen if you still think this is a required feature.

closed 2019-01-29 by bihanssen