NordicSemiconductor/pc-ble-driver-py #44: Exception on connection

gkovelman, 2017-10-24:
I'm using the pc-nrfutil v3.3.2 executable on Windows 10 which is using the ble driver v0.11.3. Connectivity chip is NRF52 on PCA10040.
I'm trying to perform secure buttonless DFU on my custom board.

The error that I'm seeing is:

> x:\nrfutil\bin>nrfutil_332.exe -v dfu ble -pkg "pkg.zip" -ic NRF52 -p COM37 -n xxx -f
> Shared library: C:\Users\Me\AppData\Local\Temp\_MEI31~1\lib\win\x86_32\pc_ble_driver_shared_sd_api_v3.dll
> Swig module name: pc_ble_driver_sd_api_v3
> Board already flashed with connectivity firmware.
> Using connectivity board at serial port: COM37
> Sending Application image.
> 
> BLE: ble_enable with local ATT MTU: 247
> BLE: Scanning for xxx
> Received advertisement report, address: 0x705681D2C705, device_name:
> Received advertisement report, address: 0x705681D2C705, device_name:
> Received advertisement report, address: 0xD7D7784BF5B7, device_name:
> Received advertisement report, address: 0xD7D7784BF5B7, device_name: xxx
> BLE: Found target advertiser, address: 0xD7D7784BF5B7, name: xxx
> BLE: Connecting to 0xD7D7784BF5B7
> Received advertisement report, address: 0xDCA904951E08, device_name:
> Received advertisement report, address: 0xD7D7784BF5B7, device_name:
> BLE: Found target advertiser, address: 0xD7D7784BF5B7, name:
> BLE: Connecting to 0xD7D7784BF5B7
> Received advertisement report, address: 0xD7D7784BF5B7, device_name: xxx
> BLE: Found target advertiser, address: 0xD7D7784BF5B7, name: xxx
> BLE: Connecting to 0xD7D7784BF5B7
> Received advertisement report, address: 0x4F351FA8E874, device_name:
> Received advertisement report, address: 0x4F351FA8E874, device_name:
> BLE: Connected to [215L, 215L, 120L, 75L, 245L, 183L]
> ATT MTU exchanged: conn_handle=0 att_mtu=247
> Successfully Connected
> BLE: Service Discovery
> Traceback (most recent call last):
>   File "nordicsemi\__main__.py", line 871, in <module>
>   File "site-packages\click\core.py", line 722, in __call__
>   File "site-packages\click\core.py", line 697, in main
>   File "site-packages\click\core.py", line 1066, in invoke
>   File "site-packages\click\core.py", line 1066, in invoke
>   File "site-packages\click\core.py", line 895, in invoke
>   File "site-packages\click\core.py", line 535, in invoke
>   File "nordicsemi\__main__.py", line 747, in ble
>   File "nordicsemi\dfu\dfu.py", line 122, in dfu_send_images
>   File "nordicsemi\dfu\dfu.py", line 83, in _dfu_send_image
>   File "nordicsemi\dfu\dfu_transport_ble.py", line 453, in open
>   File "nordicsemi\dfu\dfu_transport_ble.py", line 147, in connect
>   File "site-packages\pc_ble_driver_py\ble_driver.py", line 124, in wrapper
>   File "site-packages\pc_ble_driver_py\ble_adapter.py", line 196, in service_discovery
> TypeError: 'NoneType' object has no attribute '__getitem__'
> Failed to execute script __main__

Secure buttonless DFU is working with the iOS/Android apps/libraries, so my BLE profile is compatible.
This is also working correctly when the connectivity chip is nRF51 on PCA10031.
Log is the following:

> x:\nrfutil\bin>nrfutil_332.exe -v dfu ble -pkg "pkg.zip" -ic NRF51 -p COM38 -n xxx -f
> Shared library: C:\Users\Me\AppData\Local\Temp\_MEI14~2\lib\win\x86_32\pc_ble_driver_shared_sd_api_v2.dll
> Swig module name: pc_ble_driver_sd_api_v2
> Flashing connectivity firmware...
> Connectivity firmware flashed.
> Using connectivity board at serial port: COM38
> Sending Application image.
> BLE: Scanning for xxx
> Received advertisement report, address: 0xDCA904889157, device_name:
> Received advertisement report, address: 0xD7D7784BF5B7, device_name:
> Received advertisement report, address: 0xD7D7784BF5B7, device_name: xxx
> BLE: Found target advertiser, address: 0xD7D7784BF5B7, name: xxx
> BLE: Connecting to 0xD7D7784BF5B7
> Received advertisement report, address: 0xDCA904889157, device_name:
> Received advertisement report, address: 0xDCA904889157, device_name:
> Received advertisement report, address: 0x479528C8E156, device_name:
> BLE: Connected to [215L, 215L, 120L, 75L, 245L, 183L]
> Successfully Connected
> BLE: Service Discovery
> Bonded Buttonless characteristic discovered -> Bond
> etc...

bihanssen, 2017-10-24:
Hi,
with ATT_MTU=247 it might be that the data is being sent to the bootloader too fast, we have seen this happen before. Could you try to reduce the MTU and see if that helps? 

The pca10031 runs a softdevice that does not support MTU above 23, so that might explain why the problem is not occuring there.

gkovelman, 2017-10-24:
So fast that the service discovery fails, even before the data transmission? Weird.

I'll try to reproduce this with ATT MTU=23

bihanssen, 2017-10-24:
You're probably right, the error happens too fast for the MTU to be the issue here, unless in some indirect way.

gkovelman, 2017-10-25:
Confirming that reducing the MTU helps with the issue. I modified to `#define NRF_SDH_BLE_GATT_MAX_MTU_SIZE 23` and the PCA10040 could perform DFU.
Can we expect support for larger ATT MTU?

bihanssen, 2019-01-29:
Related to https://github.com/NordicSemiconductor/pc-nrfutil/issues/147
To be looked into.