NordicSemiconductor/pc-ble-driver-py #49: No answer from nordic chip

mohamed-elsabagh, 2018-02-22:
I flashed Nordic soft device [connectivity_2.0.1_1m_with_s132_5.0.hex](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/hex/sd_api_v5/connectivity_2.0.1_1m_with_s132_5.0.hex) into my NRF52 and I am trying to connect from my Ubuntu using Boost ver 1.63
When I run advertiser example code I can see on my logic analyzer that my Ubuntu machine is trying to send characters with baudrate 1152K but I never get response from my NRF52 I am not sure what's going on wrong here
[image]

rmcgregor1990, 2018-02-22:
That firmware uses a 1M baudrate, as the 1m in the name signifies.

mohamed-elsabagh, 2018-02-22:
sorry I miss typed the wrong firmware version, I am using the 1152K firmware and I am facing the mentioned behavior

chunfantasy, 2018-02-23:
Hi @mohamed-elsabagh , the firmware  `connectivity_2.0.1_115k2_with_s132_5.0.hex` you are using is for v5. But for the current pc-ble-driver-py is only supporting v2 and v3 now.

mohamed-elsabagh, 2018-02-23:
I apologize I used this http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832 firmware, v3 and v2 and I still have the same problem

bencefr, 2018-02-26:
In the code tree `pc-ble-driver/hex/`[sd_api_v3](https://github.com/NordicSemiconductor/pc-ble-driver/tree/99b5e78f96de60ee70e749bb89b280116275fb97/hex/sd_api_v3).

mohamed-elsabagh, 2018-02-27:
I used the mentioned firmware, but still getting the same behavior, is there a special hardware connection I should connect my nordic chip so that it can work, I am just connecting TX ,RX, DTR, CTS between nrf52 dvk and my usb2uart, is there a special pin or port should be connected to a certain level?

bihanssen, 2018-05-03:
Hi, to clarify, you have linked to documentation of two different serialization transports, Raw and HCI. Raw transport uses a two byte header, though HCI transport uses SLIP (0xC0) and a four byte header. There is no length field in HCI transport, instead SLIP is used to tell when a packets starts and ends.
pc-ble-driver uses HCI transport. 

To answer your previous question, it is correct that you need TX, RX, DTR and CTS. No other lines are needed. It is however crucial that the flow control operates correctly, or you'll quickly end up with lost bytes. Getting the hw flow control correct is often the biggest challenge when using other serial2usb converters.

mohamed-elsabagh, 2018-05-03:
so the packets sent over UART are correct, so I can't figure out where is the problem, I tried connecting CTS and RTS to ground and still can't get any response from the board, I am using BT832X from fantsel

bihanssen, 2018-05-04:
I'm not sure why things are failing with your setup. Supporting custom board setups are somewhat outside the scope of this issue tracker. I advice you to ask about your issue on devzone in addition. Also, it could be worthwhile to get pc-ble-driver running with a nordic devkit and observing how a working uart communication operates.

mohamed-elsabagh, 2018-05-15:
I lowered the optimization level on keil, flashed the firmware and run the C examples and I can now see some response from the nrf52 board, thanks

closed 2018-05-15 by mohamed-elsabagh

bihanssen, 2018-05-16:
That's good to hear. Thanks for the update.