NordicSemiconductor/pc-ble-driver-py #56: Unable to compile on Ubuntu

J-Rojas, 2018-07-06:
The master branch of pc-ble-driver compiles, however the git submodule linked in this repo gives me the follow compilation errors on Ubuntu 16.04. I'm using Boost 1.67.

```
                      ^
[ 11%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/adapter_internal.cpp.o
[ 12%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/ble_common.cpp.o
[ 12%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/ble_gap_impl.cpp.o
[ 13%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/ble_gattc_impl.cpp.o
[ 13%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/ble_gatts_impl.cpp.o
[ 14%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/ble_impl.cpp.o
[ 14%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v3.dir/src/common/sd_rpc_impl.cpp.o
In file included from /mnt/c/Users/jrojas/work/pc-ble-driver-py/pc-ble-driver/src/common/sd_rpc_impl.cpp:43:0:
/mnt/c/Users/jrojas/work/pc-ble-driver-py/pc-ble-driver/include/common/internal/transport/uart_boost.h:100:18: error: ‘io_service’ in namespace ‘boost::asio’ does not name a type
     boost::asio::io_service ioService;
                  ^
/mnt/c/Users/jrojas/work/pc-ble-driver-py/pc-ble-driver/include/common/internal/transport/uart_boost.h:102:18: error: ‘io_service’ in namespace ‘boost::asio’ does not name a type
     boost::asio::io_service::work workNotifier;
                  ^
```

olpossum, 2018-08-03:
I also encountered this issue when compiling the master branch with Boost 1.67.  I compiled Boost 1.67 on my Raspberry Pi 2 and also was able to build the master branch of pc-ble-driver.

Process:

```
echo $CMAKE_INCLUDE_PATH
/home/user/BOOST/boost_1_67_0/
sudo cmake -G "Unix Makefiles" -DCMAKE_BUILD_TYPE=release -DBOOST_LIBRARYDIR=/home/user/BOOST/boost_1_67_0/libs -DBOOST_INCLUDEDIR=$CMAKE_INCLUDE_PATH
```

Error:
```
[ 67%] Building CXX object pc-ble-driver/CMakeFiles/pc_ble_driver_obj_sd_api_v2.dir/src/common/sd_rpc_impl.cpp.o
In file included from /home/user/pc-ble-driver-py/pc-ble-driver/src/common/sd_rpc_impl.cpp:43:0:
/home/user/pc-ble-driver-py/pc-ble-driver/include/common/internal/transport/uart_boost.h:100:18: error: ‘io_service’ in namespace ‘boost::asio’ does not name a type
     boost::asio::io_service ioService;
                  ^~~~~~~~~~
/home/user/pc-ble-driver-py/pc-ble-driver/include/common/internal/transport/uart_boost.h:102:18: error: ‘io_service’ in namespace ‘boost::asio’ does not name a type
     boost::asio::io_service::work workNotifier;
```

olpossum, 2018-08-03:
I notice that ASIO was updated with BOOST 1.66.0 release.  
https://www.boost.org/users/history/version_1_66_0.html
I will try to compile 1.65.0 and see if it works...

olpossum, 2018-08-03:
different error trying with Boost 1.65.1:

```
[ 73%] Built target pc_ble_driver_obj_sd_api_v2
[ 74%] Built target pc_ble_driver_shared_sd_api_v2
[ 75%] Built target pc_ble_driver_sd_api_v2_swig_compilation
[ 76%] Linking C shared module outdir/_pc_ble_driver_sd_api_v2.so
CMakeFiles/_pc_ble_driver_sd_api_v2.dir/build.make:85: recipe for target 'outdir/_pc_ble_driver_sd_api_v2.so' failed
CMakeFiles/Makefile2:73: recipe for target 'CMakeFiles/_pc_ble_driver_sd_api_v2.dir/all' failed
Makefile:83: recipe for target 'all' failed
```

alwa-nordic, 2019-09-25:
Closing as stale. `pc-ble-driver` no longer uses Boost.

closed 2019-09-25 by alwa-nordic