NordicSemiconductor/pc-ble-driver-py #57: Failed to open. Error code: 13 when running on Raspberry Pi Zero W

xavieryin, 2018-07-17:
Trying to run on Raspberry Pi Zero W
[OS]
    latest stock RASPBIAN STRETCH LITE Version:June 2018
    setup wifi & ssh

[DK]
    PCA10028 V 1.1.0
    PCA10040 V 0.9.0

[Command Line History]
--- on PC ---
scp -pr ~/Downloads/JLink_Linux_V632h_arm/ pi@192.168.1.31:/home/pi
--- on Raspberry Pi ---
sudo cp /home/pi/JLink_Linux_V632h_arm/99-jlink.rules /etc/udev/rules.d/
sudo apt-get update
sudo apt-get dist-upgrade
mkdir /home/pi/boost
cd /home/pi/boost
wget https://dl.bintray.com/boostorg/release/1.65.1/source/boost_1_65_1.tar.bz2
tar jxvf boost_1_65_1.tar.bz2
sudo apt install cmake swig git make gcc g++ libudev-dev python python-dev python-pip
export BOOST_ROOT=/home/pi/boost/boost_1_65_1/
cd $BOOST_ROOT
./bootstrap.sh
./b2 toolset=gcc cxxflags=-fPIC cflags=-fPIC address-model=32 link=static --with-thread --with-system --with-regex --with-date_time --with-chrono
cd /home/pi
git clone https://github.com/NordicSemiconductor/pc-ble-driver-py.git
cd pc-ble-driver-py/
git submodule init
git submodule update
cd build/
cmake -G "Unix Makefiles" ..
make
pip install wrapt
cd ../python/
cp pc_ble_driver_py/examples/advertising.py .
nano advertising.py                                                 # turn off auto flasher
python ./advertising.py NRF52 /dev/ttyACM0

[Result]
pi@raspberrypi:~/pc-ble-driver-py/python $ python ./advertising.py NRF52 /dev/ttyACM0
Serial port used: /dev/ttyACM0
Traceback (most recent call last):
  File "./advertising.py", line 79, in <module>
    main(sys.argv[2])
  File "./advertising.py", line 55, in main
    driver.open()
  File "/home/pi/pc-ble-driver-py/python/pc_ble_driver_py/ble_driver.py", line 127, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

xavieryin, 2018-07-17:
By the way w/ MacBook installed pc_ble_driver_py through pip those examples runs w/o problem

xavieryin, 2018-07-18:
add information printed from BLEDriver.log_message_handler:

Serial port used: /dev/ttyACM0
Successfully opened /dev/ttyACM0. Baud rate: 115200. Flow control: none. Parity: none.

    1/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
State change: STATE_START -> STATE_RESET

    2/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       1 ->  []
               type:          RESERVED_5 reliable: no seq#:0 ack#:0 payload_length:0 data_integrity:0 err_code:0
    3/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
State change: STATE_RESET -> STATE_UNINITIALIZED

    4/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       2 ->  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       3 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
    5/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       4 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
    6/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       5 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
    7/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       6 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
    8/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       7 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
    9/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       8 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
   10/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
       9 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
   11/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
      10 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
[... 147 lines cut ...]
               [SYNC]
      36 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
      37 ->  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
State change: STATE_UNINITIALIZED -> STATE_FAILED

   36/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
   37/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
Traceback (most recent call last):
  File "/home/pi/pc-ble-driver-py/python/pc_ble_driver_py/examples/heart_rate_collector.py", line 187, in <module>
    main(serial_port)
  File "/home/pi/pc-ble-driver-py/python/pc_ble_driver_py/examples/heart_rate_collector.py", line 149, in main
    collector.open()
  File "/home/pi/pc-ble-driver-py/python/pc_ble_driver_py/examples/heart_rate_collector.py", line 67, in open
    self.adapter.driver.open()
  File "/home/pi/.local/lib/python2.7/site-packages/pc_ble_driver_py/ble_driver.py", line 127, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

olpossum, 2018-07-27:
I am having the same issue using the NRF51-DONGLE/Ubuntu 16.04 and installing pc_ble_driver_py through pip.  I also have been able to use the nrfconnect software to run serial UART application, so I know that communication with the device should be possible.

[This link](https://devzone.nordicsemi.com/f/nordic-q-a/14849/pc-ble-driver-py-driver-open-fails) references flashing /pc_ble_driver_py/hex/connectivity_115k2_with_s130_2.0.1.hex fixing the issue, however it seems that this file is no longer included with the repo.

olpossum, 2018-07-27:
I found the referenced file [here](https://github.com/NordicSemiconductor/pc-ble-driver/tree/master/hex/sd_api_v2). If you installed the pc-ble-driver from source (it looks like you did) just grab the connectivity_115k2_with_s130_2.0.1.hex file and flash it onto your device with nrfjprog.

xavieryin, 2018-07-27:
@olpossum 
git clone pc-ble-driver and you can find the binary in old commits.
Besides, If your "NRF51-DONGLE" is pca10000 then you may be hopeless since this an old-versioned nRF51 and does not support proper softdevice/SDK that connectivity firmwares use. You can take a look at compatibility matrix for more detail.

Anyway, I believe the problem is about the interface MCU since:
1. same DK w/ same nRF / interface MCU FW works on Mac
2. same DK w/ same nRF FW works when interface MCU is bypassed (Raspberry Pi directly communicates to nRF through UART), just by necessary changes to use serial port /dev/ttyS0 with HWFC

I gave up locating the root cause why DK cannot be used w/ Pi out of box since the direct connect setup fits better for my prototyping mission. It's fine with me that Nordic choose to close this issue if it doesn't attract too much interest.

xavieryin, 2018-08-20:
update status:
The open problem looks like a multi-threading timing issue. Waiting for pd-ble-driver fix to be ported for SD API V3. In the mean while, by enabling Python logging module with DEBUG level and uncomment the line in log_message_handler in ble_driver.py things gets better on my site.

bihanssen, 2019-09-25:
Closing due to inactivity, and due to the fact that pc-ble-driver-py 0.12 was released since. Consider posting on devzone if there is still an issue.

closed 2019-09-25 by bihanssen