NordicSemiconductor/pc-ble-driver-py #58: Failed to compile submodule pc-ble-driver with visual studio 2017

Seven-Zeng, 2018-08-02:
here is the build output:

**D:\Projects\pc-ble-driver-py\pc-ble-driver\build>cmake -G "Visual Studio 15 Win64" ..**
-- The C compiler identification is MSVC 19.14.26433.0
-- The CXX compiler identification is MSVC 19.14.26433.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Community/VC/Tools/MSVC/14.14.26428/bin/Hostx86/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Architecture not set, using native 64-bit toolchain.
CMake Warning (dev) at cmake/pc-ble-driver.cmake:55 (find_package):
  Policy CMP0074 is not set: find_package uses PackageName_ROOT variables.
  Run "cmake --help-policy CMP0074" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Environment variable Boost_ROOT is set to:

    D:\local\boost_1_67_0

  For compatibility, CMake is ignoring the variable.
Call Stack (most recent call first):
  CMakeLists.txt:3 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Looking for pthread.h
-- Looking for pthread.h - not found
-- Found Threads: TRUE
-- Boost version: 1.67.0
-- Found the following Boost libraries:
--   thread
--   system
--   regex
--   date_time
--   chrono
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.18.0.windows.1")
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Projects/pc-ble-driver-py/pc-ble-driver/build

**D:\Projects\pc-ble-driver-py\pc-ble-driver\build>msbuild all_build.vcxproj**
用于 .NET Framework 的 Microsoft (R) 生成引擎版本 15.7.180.61344
版权所有(C) Microsoft Corporation。保留所有权利。

生成启动时间为 2018-08-02 22:24:56。
节点 1 上的项目“D:\Projects\pc-ble-driver-py\pc-ble-driver\build\all_build.vcxproj”(默认目标)。
项目“D:\Projects\pc-ble-driver-py\pc-ble-driver\build\all_build.vcxproj”(1)正在节点 1 上生成“D:\Projects\pc-ble-driver-py\pc-ble
-driver\build\ZERO_CHECK.vcxproj”(2) (默认目标)。
PrepareForBuild:
  正在创建目录“x64\Debug\ZERO_CHECK\”。
  正在创建目录“x64\Debug\ZERO_CHECK\ZERO_CHECK.tlog\”。
InitializeBuildStatus:
  正在创建“x64\Debug\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild”，因为已指定“AlwaysCreate”。
CustomBuild:
  Checking Build System
  CMake does not need to re-run because D:/Projects/pc-ble-driver-py/pc-ble-driver/build/CMakeFiles/generate.stamp is u
  p-to-date.
FinalizeBuildStatus:
  正在删除文件“x64\Debug\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild”。
  正在对“x64\Debug\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate”执行 Touch 任务。
已完成生成项目“D:\Projects\pc-ble-driver-py\pc-ble-driver\build\ZERO_CHECK.vcxproj”(默认目标)的操作。

  **sd_rpc_impl.cpp
[... 39 lines cut ...]
ier”: 未知重写说明符 [D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\include\common\internal\transport\uart_boost.h(102): error C4430: 缺少类型说 明符 -
假定为 int。注意: C++ 不支持默认 int [D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(51): error C2065: “ioService”: 未声明的标识符 [
D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(52): error C2065: “ioService”: 未声明的标识符 [
D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(50): error C2614: “UartBoost”: 非法的成员初始化:
“ioService”不是基或成员 [D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(52): error C2614: “UartBoost”: 非法的成员初始化:
“workNotifier”不是基或成员 [D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(68): error C2065: “workNotifier”: 未声 明的标识
符 [D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(70): error C2065: “ioService”: 未声明的标识符 [
D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(70): error C2228: “.stop”的左边必须有类/结构/联合 [
D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(70): note: 类型是“unknown-type”
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(115): error C2065: “ioService”: 未声明 的标识符
[D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(178): error C2065: “ioService”: 未声明 的标识符
[D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(178): error C2228: “.stop”的左边必须有 类/结构/联合
[D:\Projects\pc-ble-driver-py\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver-py\pc-ble-driver\src\common\transport\uart_boost.cpp(178): note: 类型是“unknown-type”**

xavieryin, 2018-08-23:
which commit were you working on? looks like boost version problem.

Seven-Zeng, 2018-10-27:
@xavieryin You are right

closed 2018-10-27 by Seven-Zeng