NordicSemiconductor/pc-ble-driver-py #64: Doesn't recognize nrf52840 dongle

vrobertbaruch, 2018-09-26:
This is on Linux. While nrf_connect 2.5.0 does see the dongle, pc_ble_driver_py does not.

pip shows `enum34-1.1.6 future-0.16.0 pc-ble-driver-py-0.11.4 wrapt-1.10.11`

Here is dmesg when I plug the device in:

```
[6241327.919763] usb 3-14: new full-speed USB device number 57 using xhci_hcd
[6241328.114302] usb 3-14: New USB device found, idVendor=1915, idProduct=c00a
[6241328.114306] usb 3-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[6241328.114309] usb 3-14: Product: nRF52 Connectivity
[6241328.114312] usb 3-14: Manufacturer: Nordic Semiconductor
[6241328.114314] usb 3-14: SerialNumber: CA6A094DACD3
[6241328.115043] cdc_acm 3-14:1.1: ttyACM3: USB ACM device
```

Here's my test program, which I run under python 2.7.13:

```python
from __future__ import absolute_import
from __future__ import division
from __future__ import print_function

from pc_ble_driver_py import config
config.__conn_ic_id__ = 'NRF52'

from pc_ble_driver_py.ble_adapter import BLEAdapter
from pc_ble_driver_py.ble_driver import BLEAdvData, BLEUUIDBase, BLEUUID, BLEDriver
from pc_ble_driver_py.ble_driver import BLEEnableParams, BLEGattStatusCode
from pc_ble_driver_py.ble_driver import driver as swig_driver
from pc_ble_driver_py.ble_driver_types import list_to_char_array, uint8_array_to_list
from pc_ble_driver_py.observers import BLEAdapterObserver, BLEDriverObserver

def main():
  print('Possible dongles are at:')
  descs = BLEDriver.enum_serial_ports()
  for _, d in enumerate(descs):
    print('  {}: {} Serial Number {}'.format(d.port, d.manufacturer,
                                             d.serial_number))
  return

if __name__ == '__main__':
  main()
```

And no devices are enumerated.

Note that when I connect an nrf52 devkit, it sees the device.

bihanssen, 2018-11-22:
@vrobertbaruch, @raykamp,
thanks for reporting.
The pc-ble-driver-py does not currently have support for the nRF52840-Dongle I'm afraid. Work is in the pipeline to add support for it.

bmeisels, 2019-01-28:
@bihanssen any news on this issue?
I am working on incorporating thread dfu into a project and this could be very useful.

bihanssen, 2019-01-29:
Hi,
work on next version of pc-ble-driver-py is in development. You can follow the progress in branch feature/s132v5_py3.

sebajarosz, 2019-03-29:
Hi,
Are there any updates for the nrf52840 dongle expected in a near future? What is the plan for the support of this device?

hantianjz, 2020-01-02:
Any update on this? I am still observing the same behaviour, of not finding NRF52840 dongle, in the latest 0.14.0 version on MacOS

Krakonos, 2020-01-07:
Hi!

Actually, I just hit the same issue, but reading some posts on nordic forum lead me to believe the dongle is supported in the latest connectivity firmware and pc-ble-driver-py.

It turns out that it works pretty well, so far the only thing that seems to be broken is the enumeration of devices. As a workaround, I suggest using standard python way of find the serial port.

```
from serial.tools.list_ports import comports
def serial_ports(vendor_id: int = None, product_id: int = None):
    return [port.device for port in comports()
            if ((vendor_id is None or vendor_id == port.vid) and
                (product_id is None or product_id == port.pid))]
serial_ports(0x1915, 0xc00a)  # VID, PID of the NRF52840 dongle
```

Also note that the dongles needs to be running the USB firmware and v5 API.

I hope it helps and it gets properly sorted out soon.

bobatsar, 2020-01-10:
Hi @Krakonos,

thanks for the enumeration code.

I am also trying to get the dongle working with pc-ble-driver-py.
Can you please share which hex file you flashed and how (e.g. nrf-conenct)?
I tried connectivity_4.1.1_usb_with_s132_5.1.0.hex from pc_ble_driver_py/hex/sd_api_v5 but it didn't work.

Whats the baudrate of the connectivity FW?

Krakonos, 2020-01-10:
Sure, I'm using connectivity_4.1.1_usb_with_s132_5.1.0_dfu_pkg.zip . I think the hex might not contain the softdevice itself, as the zip I'm using has firmware and softdevice separate.

I flash using nrfutil (installed using pip), connectivity from the linux release:
```
nrfutil dfu usb-serial -p /dev/ttyACM0 -pkg /home/krakonos/src/nrf-ble-driver-4.1.1-linux_x86_64/share/nrf-ble-driver/hex/sd_api_v5/connectivity_4.1.1_usb_with_s132_5.1.0_dfu_pkg.zip
```
Note that I need to manually reset the device, as the auto-reset does not seem to work. When the nrfutil says:
```
2020-01-10 10:19:02,160 LIBUSB_ERROR_ACCESS: Unable to connect to trigger interface.
```
Just press the reset button on the dongle (note: there are two buttons, reset is the smaller one pressed from the antenna side)

For driver initialization, I'm using:
```
self.driver = BLEDriver(serial_port=device, auto_flash=False, baud_rate=1000000, log_severity_level="error")
```
.. Where device is the path to the serial port. Also, when I bypass the device discovery in the examples (for example the heart rate collector), it starts working with the dongle, so you can test it on that as well.

Note that there are some rough corners around reading/writing characterstics, especially if they are vendor specific. I have made a few pull requests to hopefully remedy those, but they are not merged (yet).

bobatsar, 2020-01-10:
Hi @Krakonos,

thanks for the fast answer.
Just to make sure. Are you using an nrf52840 Dongle? 
As far as I understand the nrf52840 needs an SD_140 for v5 and not SD_132.

[nrf52840 SD matrix](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcomp_matrix_nrf52840%2FCOMP%2Fnrf52840%2Fnrf52840_comp_matrix.html)

I tried the hex files from [the pc-ble-driver release 4.1.1](https://github.com/NordicSemiconductor/pc-ble-driver/releases/tag/v4.1.1)

Krakonos, 2020-01-10:
Interesting. I haven't found a build of api v5 and SD140, and I am under the impression that the python library supports only v3 and v5 APIs. But I still don't fully understand the softdevice and API differences between them. Regardless, it works with SD132. The link you provided even says SD132 can be used, but has limitations (albeit only for silicon revision A).

However, I can try the v6 build that has SD140. I will report back when I have time to do so.

And yes, I'm using nrf52840 dongle. Specifically the PCA10059 (but I'm not aware of any others out there). The chip itself is labeled N52840 QIAA D0 1910AC. I think QIAA is the package, 1910AC might be a date code. D0 could be silicon revision, but I haven't checked the datasheet if it's even remotely true.

bobatsar, 2020-01-10:
Ok, I tried again and realized I had at one place auto_flash=True.
Now it seems to be working with the SD_132 firmware.

Thanks a lot. Hopefully nordic will merge your PR and support also the flashing of the dongle soon.