NordicSemiconductor/pc-ble-driver-py #69: Resource leak when BLEDriver is released

MagnusWestin, 2019-01-11:
Hi
I noticed a resource leak when creating and destroying BLEDriver. There seems to be a missing call to sd_rpc_adapter_delete. The fix I did locally was to put it in BLEDriver.close. But that is of course not an ideal solution, since you then cant reuse the object if you want to. So I was thinking that moving the initialization of the following to the open function: phy_layer, link_layer, transport_layer and self.rpc_adapter.

This is how my close looks like now:
    @NordicSemiErrorCheck
    @wrapt.synchronized(api_lock)
    def close(self):
        retVal = driver.sd_rpc_close(self.rpc_adapter)
        driver.sd_rpc_adapter_delete( self.rpc_adapter )
        return retVal

If this is a good solution I can make a pull request for it. Any comments?

Magnus

bihanssen, 2019-01-29:
Hi, @MagnusWestin. Thanks for reporting. We are currently working on a new version of pc-ble-driver-py that may or may not invalidate the bug you are reporting. I will keep this ticket open for later check-up.

bihanssen, 2020-09-15:
@MagnusWestin, sorry about the long delay. If you are still dealing with ble-driver-py it would be interesting to see a PR implementing what you suggested.

MagnusWestin, 2020-09-15:
@bihanssen Unfortunately I no longer work for the same company any more. So I don't have access to the source code for the fix I did.

bihanssen, 2020-09-15:
@MagnusWestin, ok, that's understandable. I will go ahead and close this issue then as we will not look into this for now. Thank you for the suggestions.

closed 2020-09-15 by bihanssen