NordicSemiconductor/pc-ble-driver-py #112: MTU Setting on v0.12.0 fails

andrewleech, 2019-06-17:
I'm using v0.12.0 with a nrf52840 dongle running `connectivity_4.1.1_usb_with_s132_5.1.0.hex` bundled with the python package.

I'm connecting to a custom device running nrf52 with an MTU of 240, confirmed working with phone app etc.
The device normally initiates the MTU setting request.

 If I don't set the ATT / MTU in my python script, the mtu value requested by the device is overridden to 32 in `ble_adapter.BLEAdapter#on_gatts_evt_exchange_mtu_request`
``` python
    def on_gatts_evt_exchange_mtu_request(self, ble_driver, conn_handle, client_mtu):
        ble_driver.ble_gatts_exchange_mtu_reply(conn_handle, self.default_mtu)
```

The messages are truncated to 23 as expected

If I set the default_mtu in my top level class:
``` python 
class Collector(BLEDriverObserver, BLEAdapterObserver):
    def __init__(self, adapter, scenario):
        super(Collector, self).__init__()
        self.adapter.default_mtu = 240
        ...
```
The `ble_driver.ble_gatts_exchange_mtu_reply` tries to respond with 240 but fails with Error 7 (`NRF_ERROR_INVALID_PARAM`?)

If I try to initiate the MTU setting from python 
``` python
att_mtu = self.adapter.att_mtu_exchange(new_conn, 240)
```
It again fails with error 7

Is there a limitation with the bundled connectivity firmware causing these requests to fail?

andrewleech, 2019-06-17:
Note, there is also a bug in `pc_ble_driver_py.ble_driver.BLEDriver#ble_gattc_exchange_mtu_req`
https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/a9aba442c9547fe91fd2abf80bf9f27dd6c1a842/pc_ble_driver_py/ble_driver.py#L1941

The `logger.debug` block is referring to the  `att_mtu` from `ble_enable_params` doesn't exist on `nrf_sd_ble_api_ver == 5`

tkeeley79, 2019-07-31:
The MTU setting is configurable per link with `nrf_sd_ble_api_ver == 5`, so verify that the tag ID that you provide with the larger MTU setting in `sd_ble_cfg_set()` matches the tag ID that you provide with the `connect()`.   The `NRF_ERROR_INVALID_PARAM` is an indication that the MTU request with `self.adapter.att_mtu_exchange()` exceeds the MTU value set with `sd_ble_cfg_set()` for the tag ID in use with that particular connection handle.

DPMDylanPeterMiller, 2019-09-13:
I am also having issues with this.  If I have a larger MTU setup with 

`gatt_cfg.att_mtu = 276`
`gatt_cfg.tag = CFG_TAG`
`self.adapter.driver.ble_cfg_set(BLEConfig.conn_gatt, gatt_cfg)`

and when the peripheral signals a larger MTU, the central does not respond, and the driver no longer receives any notifications.

tkeeley79, 2019-09-13:
Have you tried smaller values for `gatt_cfg.att_mtu`?  247 is the maximum value for the MTU after you account for various headers and overhead in the BLE stack.

DPMDylanPeterMiller, 2019-09-15:
I tried that and got the same result.  Note that the default MTU in the "heart_rate_collector" example is 250.

closed 2019-10-18 by bihanssen

andrewleech, 2019-10-24:
Hi, thanks for the response.

I've started testing with 0.13, reflashed my nrf52840 dongle with the bundled hex `connectivity_4.1.1_usb_with_s132_5.1.0.hex` though it looks like it's the same as previous still.

I'm still getting an error with mtu of 240
```
Exception in connect_and_discover:Failed to ble_gattc_exchange_mtu_req. Error code: NRF_ERROR_INVALID_PARAM
```

This is with
``` python
    from pc_ble_driver_py.ble_adapter import BLEAdapter
    print("Creating ble driver")
    self.driver = BLEDriver(serial_port=self.serial_port, auto_flash=False)
    self.adapter = BLEAdapter(self.driver)
    self.adapter.default_mtu = 240
```

A breakpoint in `ble_adapter.BLEAdapter`:
``` python
    def on_gatts_evt_exchange_mtu_request(self, ble_driver, conn_handle, client_mtu):
        ble_driver.ble_gatts_exchange_mtu_reply(conn_handle, self.default_mtu)
```
shows `client_mtu = 240` and `self.default_mtu = 240`

But running `ble_gatts_exchange_mtu_reply` returns error 7, 'NRF_ERROR_INVALID_PARAM'

bihanssen, 2019-10-24:
@andrewleech the error you describe sounds very much a) missing config or b) not using the correct config tag in the call to connect(). See @tkeeley79's comment on this: https://github.com/NordicSemiconductor/pc-ble-driver-py/issues/112#issuecomment-517043302
Please look over the code and make sure the same tag is used in cfg_set and connect, and that the tag is not 0 because that uses default setup values.

andrewleech, 2019-10-24:
Thanks for the suggestions, though I'm not sure that's it. 
If I don't change self.default_mtu then the mtu stays at 23  and the rest of our app works fine, we authenticate, bond, read and write characteristics. We obviously just get the first 23 bytes, rather than the full message our device is trying to send.

I'm trying to communicate with a custom device based on a nrf5280, the device works fine with nrf connect on desktop and phones (mtu 240 fine then) it's only with this library I'm having a problem.

I tried compiling my own connectivity firmware from the latest SDK, but this library can't connect/communicate with the figure at all then.
Same of I let nrf connect on PC reflash the dongle, then this library can't talk to the dongle.

Do you have any example code you used to configure and use a larger mtu?
Do you know what nrf hardware & firmware was used for the test?

andrewleech, 2020-08-05:
Hi @bihanssen, I've only just got back to this project and tested again with the v14.2 release. 

I'm still using a nRF52840 dongle (same model but different actual one from last time).

I flashed it with `nrfutil dfu usb-serial -p COM14 -pkg C:\py3.8\lib\site-packages\pc_ble_driver_py\hex\sd_api_v5\connectivity_4.1.1_usb_with_s132_5.1.0_dfu_pkg.zip` included in the release.

I ran the new `test_mty.py` (thanks for that) but with minimal modification to only open one adapter as central and connect by name to my custom hardware. 
This custom hardware uses an mtu of 244 normally which works when communicating with a phone or PC running nrf connect.

`python tests\test_mtu.py --nrf-family NRF52 --log-level info --mtu 244 --port-a COM13`

Again, I got the same failure as previously:
```
2020-08-05 14:37:16,235 [16536/LogThread] Successfully opened COM13. Baud rate: 1000000. Flow control: none. Parity: none.
2020-08-05 14:37:17,612 [32392/MainThread] Central adapter is 2432117281712
2020-08-05 14:37:17,612 [32392/MainThread] scan_start, trying to find LCR
2020-08-05 14:37:19,672 [27104/EventThread] Trying to connect to peripheral advertising as LCR, address: 0x023730181DE6
2020-08-05 14:37:20,005 [27104/EventThread] Exception: MTU exchange reply failed. Common causes are: missing att_mtu setting in ble_cfg_set, different config tags used in ble_cfg_set and adv_start.
2020-08-05 14:37:20,007 [27104/EventThread] <FrameSummary file C:\py3.8\lib\site-packages\pc_ble_driver_py\ble_driver.py, line 2605 in ble_event_handler_sync>
2020-08-05 14:37:20,007 [27104/EventThread] <FrameSummary file C:\py3.8\lib\site-packages\pc_ble_driver_py\ble_adapter.py, line 707 in on_gatts_evt_exchange_mtu_request>
2020-08-05 14:37:20,007 [27104/EventThread] 
2020-08-05 14:37:20,037 [16536/LogThread] serial port COM13 closed.
```
A debugger stepping through showed that when the `Exception: MTU exchange reply failed` is thrown the underlying failure is still `NRF_ERROR_INVALID_PARAM`.

I'll order another dongle so I can run the test unmodified, though I fail to see why this should matter at this point.

What hardware are you using when you run these tests? 
Are you using the same firmware zip as I have flashed?

bihanssen, 2020-08-10:
@andrewleech, the firmware you are using is correct, so that should not be the issue.

The test fails in the exchange reply. This might indicate that the peer device (your custom device) does not accept the requested MTU. Not that the MTU request in the test is hardcoded to 150. The --mtu command line option is overridden in this test. Maybe that is the reason for the failure, is your device does not accept MTU 150?

You could try setting log level to debug to see if that reveals some more clues.