NordicSemiconductor/pc-ble-driver-py #115: pc_ble_driver_py DLL missing

konovagit, 2019-07-08:
I'm having an issue running the examples provided within the pc_ble_driver_py repository.

When I try running an example, say, advertising.py,  I get the following error: ImportError: DLL load failed

I followed the instructions Installing from PyPl specified in https://github.com/NordicSemiconductor/pc-ble-driver-py .

The DLL being loaded is loaded in pc_ble_driver_py/lib/nrf_ble_driver_sd_api_v5.py, line 13: from . import _nrf_ble_driver_sd_api_v5

Version 0.12.0 on Windows 10 Pro
Ref but no response: https://devzone.nordicsemi.com/f/nordic-q-a/48489/pc_ble_driver_py-dll-missing

kenr, 2019-07-09:
Can you provide the first line printed after starting python REPL?
For example:
```
c:\>python
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32
```

konovagit, 2019-07-09:
Thank you for your reply.

My python settings :
```
C:\Users\konova>python3
Python 3.7.2 (tags/v3.7.2:9a3ffc0492, Dec 23 2018, 23:09:28) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
```

marknolan, 2019-07-09:
Same issue for me (as posted here: https://devzone.nordicsemi.com/f/nordic-q-a/47957/pc-ble-python-py-v0-12-0). We got it working on a second PC in the office for the same project so there is no conflict with other libraries.

My python REPL is:
Python 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 22:22:05) [MSC v.1916 64 bit (AMD64)] on win32

konovagit, 2019-07-10:
Response of Michael Portmann from Nordic dev zone:

> > I found the same problem today.
> 
> I used Dependency Walker to examine the DLL provided with the pc-ble-driver-py package for 0.12.0, I think the supplied DLLs have missing dependencies that may be included with Visual Studio (Debug build?)
> 
> The change notes for pc-ble-driver-py 0.12.0 mention use of the latest driver 4.1.1, if you go there pc-ble-driver 4.1.1 you can download it and those DLL files do not have the same dependencies, at least on my machine.
> 
> I replaced pc_ble_driver_py.lib.nrf-ble-driver-sd_api_v5-mt-gd-4_1_1.dll with a copy of nrf-ble-driver-sd_api_v5-mt-4_1_1.dll and now my installation works.

Works for me !

Suhail-Ahamed, 2019-07-25:
By replacing that also I still facing the same issue! Did you faced it initially and figured out anything later thus it working now?

jupe, 2019-10-03:
> I replaced pc_ble_driver_py.lib.nrf-ble-driver-sd_api_v5-mt-gd-4_1_1.dll with a copy of nrf-ble-driver-sd_api_v5-mt-4_1_1.dll and now my installation works.

That helps for me also. Very annoying bug. Please provide correct dll files for windows, otherwise this causes a lot of headache !

jupe, 2019-10-16:
seems that v0.13.0 fixes most of issues with older version, including this one.

closed 2019-10-16 by kenr