NordicSemiconductor/pc-ble-driver-py #121: Application Error response for Write Request not supported

sw, 2019-09-24:
BLE specifies that error codes 0x80 – 0xFF are for vendor-specific application errors. (Core 4.0, Volume 3, Part F, Section 3.4.1.1)

This doesn't work because ble_driver.py tries to cast the value to `BLEGattStatusCode` when handling `BLEEvtID.gattc_evt_write_rsp`. That enum does not define the values in the above range.

I'm trying to send error code 0xA1 from a Peripheral, to pc-ble-driver-py acting as Central. I get the following error message:
`Exception: 417 is not a valid BLEGattStatusCode`

417 is 0x1A1, I don't really see where the 0x100 comes from.

alwa-nordic, 2019-09-25:
We are looking at a fix for the missing enum values in PR #122.

As for the 0x100, feel free to open a separate issue. But since it is public API, it will most likely be a wontfix or notabug.

closed 2019-09-25 by alwa-nordic

sw, 2019-09-28:
Thanks for the quick fix @alwa-nordic !

My code is working fine now, though I had to wrap the call to `write_req` in a try...catch to get the error code. This is because the error codes are caught by the `NordicSemiErrorCheck` wrapper and not returned from the function, but raise an exception. That probably makes sense for the common case where you don't expect an error.

To anyone looking at this, the offset of 0x100 simply comes from the definition of `BLE_GATT_STATUS_ATTERR_..` codes in the softdevice.