NordicSemiconductor/pc-ble-driver-py #148: 0.14.0 requires pip 19.0 or higher

carlescufi, 2019-12-06:
When you build the wheels for version 0.14.0, you are using the [manylinux2010](https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/7c551e24d9f527a398ae85e94b16b4c24e668266/azure-pipelines.yml#L9) platform tag. 
Support for this tag was introduced in [pip 19.0](https://pip.pypa.io/en/stable/news/#id152) but even the most recent Ubuntu (19.10) ships with pip 18.1.

Could you consider building for both `manylinux1` and `manylinux2010` so that older pip versions can install this wheel?

bihanssen, 2019-12-10:
Thanks for the detailed report @carlescufi. We will look into if manylinux1 will work in place of manylinux2010. If manylinux1 works fine, I don't think there's a need to keep manylinux2010 in addition.

mrtzcspr, 2020-01-03:
Hi, I used the virtualenv workaround from @carlescufi . Not sure if this is related to the workaround..

I'm running this in a docker container with ubuntu:18.04
```
nrfutil settings generate --family NRF52 --application "${APP_HEX}" --application-version ${APP_INDEX_INTEGER} --bootloader-version ${BOOT_INDEX_INTEGER} --bl-settings-version 2 --app-boot-validation VALIDATE_GENERATED_CRC --sd-boot-validation VALIDATE_GENERATED_CRC --softdevice "${SOFTDEVICE_HEX}" "${BOOT_SETTINGS_HEX}"
```

And now I get this error. This worked before with a slightly older version of nrfutil installed with the normal ubuntu pip3
```
Traceback (most recent call last):
  File "/opt/venv/bin/nrfutil", line 5, in <module>
    from nordicsemi.__main__ import cli
  File "/opt/venv/lib/python3.6/site-packages/nordicsemi/__main__.py", line 52, in <module>
    from nordicsemi.dfu.dfu_transport_serial import DfuTransportSerial
  File "/opt/venv/lib/python3.6/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 53, in <module>
    from nordicsemi.dfu.dfu_trigger import DFUTrigger
  File "/opt/venv/lib/python3.6/site-packages/nordicsemi/dfu/dfu_trigger.py", line 48, in <module>
    import usb1
  File "/opt/venv/lib/python3.6/site-packages/usb1/__init__.py", line 61, in <module>
    from . import libusb1
  File "/opt/venv/lib/python3.6/site-packages/usb1/libusb1.py", line 199, in <module>
    libusb = _loadLibrary()
  File "/opt/venv/lib/python3.6/site-packages/usb1/libusb1.py", line 173, in _loadLibrary
    return dll_loader('libusb-1.0' + suffix, **loader_kw)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libusb-1.0.so: cannot open shared object file: No such file or directory
```

Installing the apt-package `libusb-dev` doesn't seam to resolve it.

carlescufi, 2020-01-09:
@mrtzcspr this looks like you are missing libusb, which is unrelated I assume. Perhaps @bihanssen knows more about this.

mrtzcspr, 2020-01-09:
@carlescufi I think my problem was caused by this issue: https://github.com/NordicSemiconductor/pc-nrfutil/issues/295
Because my docker setup had no issue when I used `pip install nrfutil==6.0.0`.

bihanssen, 2020-01-09:
It’s likely related to issue 295 as stated. It is fixed in master, to be released shortly.

floanwelt, 2020-01-27:
@bihanssen I just ran into the same problem running on a device in which I do not want to update to a current pip version for compatibility reasons. Did you have a chance to check if manylinux1 will work and if you'll release a new wheel for it soon?

bihanssen, 2020-02-10:
@floanwelt we have tried building with manylinux1 and unfortunately it was not possible. Some of the tool versions provided in manylinux1 are simply too old for compiling pc-ble-driver-py.

Closing this isssue. The requirement for pip 19.0 still stands as we have not found a good way to resolve the issue.

closed 2020-02-10 by bihanssen

Xenoamor, 2020-03-06:
This fails for me on alpine with pip-20.0.2 with the exact same error. Using nrfutil==5.2.0 installs fine but anything about that doesn't.
For some reasons pip3 just won't find any pc_ble_driver_py versions beyond 0.11.4

jupe, 2020-03-06:
I had similar issue with alpine container, pip3 can’t find pc-ble-driver-py than 0.11.4 and earlier versions. with ubuntu also latest releases found and installation success. I would like to use alpine instead.

carlescufi, 2020-03-09:
> This fails for me on alpine with pip-20.0.2 with the exact same error. Using nrfutil==5.2.0 installs fine but anything above that doesn't.
> 
> For some reason pip3 just won't find any pc_ble_driver_py versions beyond 0.11.4

Are you completely sure you are using pip3 v2.0.0.2? Could you please attach a full log of the error when running `pip3 -vvv install --user nrfutil` while making sure first that you have uninstalled both the `nrfutil` and `pc-ble-driver` packages first? Also please attach the output of `pip3 --version`.

Xenoamor, 2020-03-09:
[I've made a pastebin for the output here](https://pastebin.com/hd846H3t)

The dockerfile I'm using to generate this is:
```
FROM alpine:3.11

# Install nrfutil
RUN apk --no-cache add python3 py3-pip
RUN pip3 install --upgrade pip # As it installs an older version by default
RUN pip3 --version
RUN pip3 -vvv install --user nrfutil```

lchish, 2020-09-17:
It's failing for me using raspbian 10.4.
Here's the relevant part of the logs https://pastebin.com/4CrR79Rr
It's checking two repositories for version info 
* https://pypi.org/simple/pc-ble-driver-py/ 
* https://www.piwheels.org/simple/pc-ble-driver-py/
The first doesn't have an ARM build so it skips it
The second is out of date, the most recent build being 0.11.4.

So it seems the issue is that the piwheels repo is out of date.