NordicSemiconductor/pc-ble-driver-py #153: The ble_adapter.py read_req doesn't allow reading data > ATT MTU

RyanLindemanCAE, 2020-02-10:
I am using the pc-ble-driver-py as a platform to test my BLE peripheral device. I need the ability to read characteristics which exceed the ATT MTU value. This is possible if an offset parameter is added to the read_req method found in ble_adapter.py and then using a similar algorithm to the following:
    def read(self, uuid, base=None):
        # Use read_req to get first few bytes (offset=0)
        status, data = self._read_req(uuid, base=base)
        if not status:
            return None
        # If data filled maximum size allowed there might be more data
        elif len(data) == (self.att_mtu - 1):
            # Read more while status is OK and length is not 0
            message = []
            while status and len(data):
                if data:
                    message.extend(data)
                status, data = self._read_req(uuid, len(message), base=base)
        # Otherwise, just return the short data received
        else:
            return return data
        return message

    def _read_req(self, uuid, offset=0, base=None):
        status, data = self.controller.adapter.read_req(self.conn_handle, BLEUUID(uuid, base=base), offset=offset)
        return status == BLEGattStatusCode.success, data

RyanLindemanCAE, 2020-03-05:
Thank you for merging my pull request that addresses this issue. Do you have a timeline when a release will be made that includes this fix?

bihanssen, 2020-03-09:
The code was part of v0.14.2 that was released to pypi a while ago (Feb 14th), though it hadn't been released here on github until now. Closing the issue.

closed 2020-03-09 by bihanssen