NordicSemiconductor/pc-ble-driver-py #158: The ble_adapter.conn_param_update raises exception when used due to missing status

RyanLindemanCAE, 2020-03-05:
While troubleshooting why my Peripheral device couldn't stay connected for longer than ~60 seconds I discovered that my Python Central based on the pc-ble-driver-py module was not responding to an incoming Connection Parameter Request from the Peripheral. In adding code (which was missing in the all the examples) to my Python Central I first attempted to use the ble_adapter.conn_param_update method to send a response to the Peripheral. Here is the current implementation:
`     @NordicSemiErrorCheck(expected=BLEGattStatusCode.success)
    def conn_param_update(self, conn_handle, conn_params):
        self.driver.ble_gap_conn_param_update(conn_handle, conn_params)
        result = self.evt_sync[conn_handle].wait(evt=BLEEvtID.gap_evt_conn_param_update)
        return result["status"]
`
The issue is that the BLEEvtID.gap_evt_conn_param_update event does not provide a "status" element in the result object returned. This causes an invalid index exception to be raised on the return result["status"] line shown above. As a workaround, I ended up calling the ble_driver.ble_gap_conn_param_update method instead and just ignoring any response returned by the Peripheral but the better solution would be to fix this seemingly unused method (and if possible provide better examples that actually read/write real characteristics as a Central device). It would also be preferable if the Python Central example included performing this call at appropriate times.

bihanssen, 2020-03-09:
@RyanLindemanCAE thanks for the report. Marking as bug for follow up.

jupe, 2020-04-06:
We have faced this issue also. Correction would be appreciated.

bihanssen, 2020-04-06:
I added a test and fix proposal in PR #162. Not merged yet.

bihanssen, 2020-04-30:
Fixed by #162

closed 2020-04-30 by bihanssen