NordicSemiconductor/pc-ble-driver-py #165: nrf connect dongle failed to run the example centrality

SYJshang, 2020-05-07:
When I was running the heart rate example, I would report this error below, someone tell me what caused this error, which causes the driver to fail to recognize now. But I found the device on the NRF connect desktop and on the command line.
```
python3 heart_rate_collector.py NRF52 /dev/tty.usbmodemC8A5C57A42372 ['heart_rate_collector.py', 'NRF52', '/dev/tty.usbmodemC8A5C57A42372'] Serial port used: /dev/tty.usbmodemC8A5C57A42372 2020-05-07 16:40:30,680 [123145336700928/LogThread] Successfully opened /dev/tty.usbmodemC8A5C57A42372. Baud rate: 1000000. Flow control: none. Parity: none. 2020-05-07 16:40:30,680 [123145336700928/LogThread] evt> severity(20) message(Successfully opened /dev/tty.usbmodemC8A5C57A42372. Baud rate: 1000000. Flow control: none. Parity: none.) 2020-05-07 16:40:30,681 [123145341956096/StatusThread] RpcAppStatus.resetPerformed: Target Reset performed 2020-05-07 16:40:30,681 [123145341956096/StatusThread] evt> status code(RpcAppStatus.resetPerformed) message(Target Reset performed) 2020-05-07 16:40:31,889 [123145341956096/StatusThread] RpcAppStatus.connectionActive: Connection active 2020-05-07 16:40:31,890 [123145341956096/StatusThread] evt> status code(RpcAppStatus.connectionActive) message(Connection active) 2020-05-07 16:40:31,995 [123145341956096/StatusThread] RpcAppStatus.ioResourcesUnavailable: serial port read failed on port /dev/tty.usbmodemC8A5C57A42372. Error: Device not configured [6] 2020-05-07 16:40:31,995 [123145341956096/StatusThread] evt> status code(RpcAppStatus.ioResourcesUnavailable) message(serial port read failed on port /dev/tty.usbmodemC8A5C57A42372. Error: Device not configured [6]) 2020-05-07 16:40:33,391 [123145336700928/LogThread] Failed to receive response for command Traceback (most recent call last): File "heart_rate_collector.py", line 222, in 2020-05-07 16:40:33,392 [123145336700928/LogThread] evt> severity(30) message(Failed to receive response for command) 2020-05-07 16:40:33,393 [123145341956096/StatusThread] RpcAppStatus.pktSendError: Error sending packet to target. Code: 0x8016 2020-05-07 16:40:33,393 [123145341956096/StatusThread] evt> status code(RpcAppStatus.pktSendError) message(Error sending packet to target. Code: 0x8016) main(serial_port) File "heart_rate_collector.py", line 178, in main collector.open() File "heart_rate_collector.py", line 103, in open self.adapter.driver.ble_cfg_set(BLEConfig.conn_gatt, gatt_cfg) File "/usr/local/lib/python3.7/site-packages/pc_ble_driver_py/ble_driver.py", line 107, in wrapper error_code=err_code, pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_cfg_set. Error code: NRF_ERROR_SD_RPC_SEND
```

bihanssen, 2020-05-08:
Hi, it looks like you are using an incompatible connectivity firmware on the device. You'll need a connectivity that uses SoftDevice 5. If you have programmed the device with BLE app in nRF Connect for Desktop you will have SoftDevice 3, which would explain the error message. Correct connectivity hex files are in pc_ble_driver_py/hex folder.

SYJshang, 2020-05-08:
我测试一下。

closed 2020-05-13 by SYJshang