NordicSemiconductor/pc-ble-driver-py #171: Failed to ble_gap_data_length_update. Error code: NRF_ERROR_RESOURCES

zhangzhii, 2020-09-01:
Hi, I'm using a v5 nrf52840 dongle to communicate with a custom nrf52832 device.  

Data_length requested by the device is 251. While using nrf_connect, data_length is set to be 251 automatically. But if I don't set data_length in my script, the value is cut down to 151:
```
2020-09-01 16:41:07,671 - device.ble - INFO - on_gap_evt_connected - conn_handle=0, peer_addr=<pc_ble_driver_py.ble_driver.BLEGapAddr object at 0x000002890AA7CCC0>, role=BLEGapRoles.central, conn_params=Connection Parameters min_conn_interval_ms(30.0) max_conn_interval_ms(30.0) conn_sup_timeout_ms(4000.0) slave_latency(0)
2020-09-01 16:41:07,671 - gui.gui - INFO - Connected
2020-09-01 16:41:07,680 - device.ble - INFO - on_gatts_evt_exchange_mtu_request: ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x000002890BC88160>, conn_handle=0, client_mtu=247
2020-09-01 16:41:07,705 - device.ble - INFO - on_gap_evt_data_length_update_request - ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x000002890BC88160>, conn_handle=0, max_tx_octets=251, max_rx_octets=251, max_tx_time_us=2120, max_rx_time_us=2120
2020-09-01 16:41:07,732 - device.ble - INFO - on_gap_evt_data_length_update - ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x000002890BC88160>, conn_handle=0, max_tx_octets=151, max_rx_octets=151, max_tx_time_us=1320, max_rx_time_us=1320
2020-09-01 16:41:09,952 - device.ble - INFO - Battery level = [43], status = BLEGattStatusCode.success
```

I tried to set data_length but got `NRF_ERROR_RESOURCES`.
```
    def on_gap_evt_data_length_update_request(self, ble_driver, conn_handle, data_length_params):
        logger.info("on_gap_evt_data_length_update_request - ble_driver={}, conn_handle={}, max_tx_octets={}, max_rx_octets={}, max_tx_time_us={}, max_rx_time_us={}".format(ble_driver, conn_handle, data_length_params.max_tx_octets, data_length_params.max_rx_octets, data_length_params.max_tx_time_us, data_length_params.max_rx_time_us))
        self._adapter.data_length_update(conn_handle, 251)
        # self._adapter.driver.ble_gap_data_length_update(conn_handle, data_length_params, None)
```
result:
```
2020-09-01 17:18:27,080 - device.ble - INFO - on_gatts_evt_exchange_mtu_request: ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x0000029326702A58>, conn_handle=0, client_mtu=247
2020-09-01 17:18:27,110 - device.ble - INFO - on_gap_evt_data_length_update_request - ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x0000029326702A58>, conn_handle=0, max_tx_octets=251, max_rx_octets=251, max_tx_time_us=2120, max_rx_time_us=2120
2020-09-01 17:18:27,113 - pc_ble_driver_py.ble_driver - ERROR - Exception: Failed to ble_gap_data_length_update. Error code: NRF_ERROR_RESOURCES
2020-09-01 17:18:27,241 - pc_ble_driver_py.ble_driver - ERROR - <FrameSummary file D:\Anaconda3_64\envs\Crimson\lib\site-packages\pc_ble_driver_py\ble_driver.py, line 2650 in ble_event_handler_sync>
2020-09-01 17:18:27,241 - pc_ble_driver_py.ble_driver - ERROR - <FrameSummary file D:\BrainCo\Documents\crimson_mini_python_sdk\crimson_ble_tool\device\ble.py, line 284 in on_gap_evt_data_length_update_request>
2020-09-01 17:18:27,242 - pc_ble_driver_py.ble_driver - ERROR - <FrameSummary file D:\Anaconda3_64\envs\Crimson\lib\site-packages\pc_ble_driver_py\ble_adapter.py, line 242 in data_length_update>
2020-09-01 17:18:27,242 - pc_ble_driver_py.ble_driver - ERROR - <FrameSummary file D:\Anaconda3_64\envs\Crimson\lib\site-packages\pc_ble_driver_py\ble_adapter.py, line 240 in data_length_update>
2020-09-01 17:18:27,242 - pc_ble_driver_py.ble_driver - ERROR - <FrameSummary file D:\Anaconda3_64\envs\Crimson\lib\site-packages\pc_ble_driver_py\ble_driver.py, line 107 in wrapper>
2020-09-01 17:18:27,242 - pc_ble_driver_py.ble_driver - ERROR - 
2020-09-01 17:18:27,244 - device.ble - INFO - on_gap_evt_data_length_update - ble_driver=<pc_ble_driver_py.ble_driver.BLEDriver object at 0x0000029326702A58>, conn_handle=0, max_tx_octets=151, max_rx_octets=151, max_tx_time_us=1320, max_rx_time_us=1320
2020-09-01 17:18:29,355 - device.ble - INFO - Battery level = [34], status = BLEGattStatusCode.success
```

What is the correct way to set data_length?
Thanks in advance.

bihanssen, 2020-09-01:
You could refer to the test https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/tests/test_data_length.py, it uses data length up to 251. Note that you'll need to modify event_length to get the higher data lenghts.

zhangzhii, 2020-09-02:
It works after I set event_length to 5! 
Thank you!!

closed 2020-09-02 by zhangzhii

bihanssen, 2020-09-02:
Good to hear, thanks for the update.