NordicSemiconductor/pc-ble-driver-py #182: Python 3.9 support

libcg, 2021-01-12:
https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/setup.py#L122 needs to be updated for the new Python version.

FlyingBBQ, 2021-01-19:
Note that simply updating that line to support Python 3.9+, for example `python_requires=">=3.6"` causes problems. Running this driver with Python 3.9 results in segfaults. Apparently there are incompatible changes in Python 3.9.

It would be great if this driver can be updated to support the latest version of Python. Currently I need to manually install an older Python version (3.8) and create a `venv` to use this driver for my test scripts.

andrew-gillan, 2021-01-19:
A new parser was introduced in 3.9. It may be worth checking if using the
old parser "fixes" your issue in the meantime.

See https://docs.python.org/3.9/whatsnew/3.9.html#new-parser for how to use
the old parser.

katjie, 2021-05-11:
I can see that there is a commit that provides Python 3.9 support,
https://github.com/NordicSemiconductor/pc-ble-driver-py/commit/303127e2b8e24d72d3d209c505d637c755bb3d42
but no official release. Is there a timeline for when we can expect an official release of a version with the Python 3.9 support?

br101, 2021-05-25:
Yes please, need that for ubuntu 21.04

snoonan-wp, 2021-07-10:
+1 for new release to Pypi

katjie, 2021-07-10:
@snoonan-wp according to [this devzone ticket](https://devzone.nordicsemi.com/f/nordic-q-a/75234/release-plan-for-pc-ble-driver-py-python-3-9-support) they are working on a release. There was some hope it would have been released a couple of weeks ago, so I'm hoping that it will come soon.

katjie, 2021-10-06:
I think this issue can be closed, since 0.16.2 has been released to pypi.
Of course the next question is about python 3.10 support, but that can be another issue :o)

br101, 2021-10-06:
Yes, works for me now :+1: