NordicSemiconductor/pc-ble-driver-py #185: PyPI source download not available

cederom, 2021-01-27:
Hello, I am working on FreeBSD, there is no source code download on PyPI for `pc-ble-driver-py`, that makes it unavailable to install with `pip` on platforms for which WHL files are not provided. This package is a hard dependency for https://github.com/NordicSemiconductor/pc-nrfutil project.

https://pypi.org/project/pc-ble-driver-py/0.15.0/#files

The last release that has source code on PyPI is 0.11.4 and that provides `nrfutil` that is incompatible with Python 3.

Please also provide source code download on PyPI for every release of `pc-ble-driver-py` as well as `nrfutil`.

LimeSlice, 2022-05-27:
This is still true. Can this be updated with PyPi?

dvzrv, 2022-08-01:
Hi! I'm packaging this project for Arch Linux. Unfortunately I am not able to rely on sdist tarballs from pypi.org.
It would be great if they could be added!