NordicSemiconductor/pc-ble-driver-py #215: Breaks on MacOS Monterey, hardcoded library location

pavelkomarov, 2022-05-05:
You guys hardcoded your library to look for python in a location where it's no longer installed, as of Monterey.

```
ImportError: dlopen(.../lib/python3.9/site-packages/pc_ble_driver_py/lib/_nrf_ble_driver_sd_api_v5.so, 0x0002): Library not loaded: /Library/Frameworks/Python.framework/Versions/3.9/Python
  Referenced from: .../lib/python3.9/site-packages/pc_ble_driver_py/lib/_nrf_ble_driver_sd_api_v5.so
  Reason: tried: '/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file)
```

I can work around the issue by creating a symlink

```
ln -s /usr/local/Cellar/python@3.9/3.9.7/Frameworks/Python.framework/Versions/3.9/ /Library/Frameworks/Python.framework/Versions/3.9
```

but I shouldn't have to. This should be fixed to work with python installs in the new location.

jupe, 2022-11-16:
This leads several workarounds & extra documentation around projects that based on pc-ble-driver-py so fix would be more than welcome. I guess this should not need much effort overall..

savvn001, 2022-12-12:
I have the same issue except it's looking for Python version 2.7 in system directory:

```
ImportError: dlopen(/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pc_ble_driver_py/lib/macos_osx/_pc_ble_driver_sd_api_v3.so, 0x0002): Library not loaded: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python'
  Referenced from: '/Library/Frameworks/Python.framework/Versions/3.11/lib/python3.11/site-packages/pc_ble_driver_py/lib/macos_osx/_pc_ble_driver_sd_api_v3.so'
  Reason: tried: '/System/Library/Frameworks/Python.framework/Versions/2.7/Python' (no such file)
```

Why is yours looking for 3.9? I can't symlink as 2.7 isn't installed on my system